earlephilhower/mklittlefs #12: macOS build does not work

007hacky007, 2020-11-28:
macOS build does not work on macOS 10.15.7 (Catalina)

Not sure whether it is expected to work or not and for which version of macOS the build is intended for - but I just wanted to share this as maybe someone else may experience this error as well.

```
$ tar xfv x86_64-apple-darwin14-mklittlefs-47e8167.tar.gz
x mklittlefs/
x mklittlefs/mklittlefs
$ cd mklittlefs
$ ./mklittlefs
[1]    33495 abort      ./mklittlefs
```

Building from the source fortunately works without any issue
```
$ make dist
c++ -std=gnu++11 -Os -Wall   -Itclap -Iinclude -Ilittlefs -I. -D VERSION=\"0.2.3-31-g295fe9b\" -D LITTLEFS_VERSION=\"v2.0.5\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__ -D LFS_NAME_MAX=32   -c -o main.o main.cpp
cc -std=gnu99 -Os -Wall   -Itclap -Iinclude -Ilittlefs -I. -D VERSION=\"0.2.3-31-g295fe9b\" -D LITTLEFS_VERSION=\"v2.0.5\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__ -D LFS_NAME_MAX=32   -c -o littlefs/lfs.o littlefs/lfs.c
cc -std=gnu99 -Os -Wall   -Itclap -Iinclude -Ilittlefs -I. -D VERSION=\"0.2.3-31-g295fe9b\" -D LITTLEFS_VERSION=\"v2.0.5\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__ -D LFS_NAME_MAX=32   -c -o littlefs/lfs_util.o littlefs/lfs_util.c
c++ main.o littlefs/lfs.o littlefs/lfs_util.o -o mklittlefs
strip mklittlefs
cp mklittlefs mklittlefs-0.2.3-31-g295fe9b-generic-osx/
tar czf mklittlefs-0.2.3-31-g295fe9b-generic-osx.tar.gz mklittlefs-0.2.3-31-g295fe9b-generic-osx
make: *** No rule to make target `test', needed by `dist'.  Stop.

$ cd mklittlefs-0.2.3-31-g295fe9b-generic-osx
$ ./mklittlefs
PARSE ERROR:
             Required arguments missing: list, unpack, create, image_file

Brief USAGE:
   ./mklittlefs  {-c <pack_dir>|-u <dest_dir>|-l} [-d <0-5>] [-a] [-b
                 <number>] [-p <number>] [-s <number>] [--] [--version]
                 [-h] <image_file>

For complete USAGE and HELP type:
   ./mklittlefs --help
```

earlephilhower, 2020-11-28:
Actually, all "make dist"s would crap out due to an obsolete "tests" target.  Just removed it, "make dist" should be nice and clean now.

closed 2020-11-28 by earlephilhower

007hacky007, 2020-11-28:
@earlephilhower great! But will this change fix the broken darwin release on the releases page https://github.com/earlephilhower/mklittlefs/releases ?

earlephilhower, 2020-11-28:
The makefile was just trying to build tests, which were not added to the repo ever (this was a fork from the mkspiffs app).  So no binary changes were done to the app, just the makefile won't throw an error after making the tarball.

For MacOS, I'm using the Xcode cross compiler on Linux and it's been fine AFAIK since we have had many Mac users on the Arduino core which uses this (and the cross-compiled GCC toolchain, too).

Is this related to the new OSX release?  I don't use a Mac, but I read about it briefly.  It looks like it's only allowing Apple signed/approved executables by default, but I assume there's a switch to allow unsigned binaries.  If not, Apple has basically killed the entire OSS ecosystem for the Mac. :(

007hacky007, 2020-11-28:
Well, I'm not using the newly released macOS, but macOS Catalina which has been around for 1 year already... So I would expect it to be working. When I try to start it, it exits immediately with the "SIGABRT".

```
$  ./mklittlefs
[1]    37023 abort      ./mklittlefs
```

```
$ sudo dtruss ./mklittlefs
dtrace: system integrity protection is on, some features will not be available

SYSCALL(args) 		 = return
open("/dev/dtracehelper\0", 0x2, 0xFFFFFFFFEFBFEDA0)		 = 3 0
ioctl(0x3, 0x80086804, 0x7FFEEFBFECB0)		 = 0 0
close(0x3)		 = 0 0
access("/AppleInternal/XBS/.isChrooted\0", 0x0, 0x0)		 = -1 2
bsdthread_register(0x7FFF6A5D0B7C, 0x7FFF6A5D0B68, 0x2000)		 = 1073742047 0
sysctlbyname(kern.bootargs, 0xD, 0x7FFEEFBFDE80, 0x7FFEEFBFDE70, 0x0)		 = 0 0
issetugid(0x0, 0x0, 0x0)		 = 0 0
ioctl(0x2, 0x4004667A, 0x7FFEEFBFE1C4)		 = 0 0
mprotect(0x100129000, 0x1000, 0x0)		 = 0 0
mprotect(0x100130000, 0x1000, 0x0)		 = 0 0
mprotect(0x100131000, 0x1000, 0x0)		 = 0 0
mprotect(0x100138000, 0x1000, 0x0)		 = 0 0
mprotect(0x100139000, 0x1000, 0x0)		 = 0 0
mprotect(0x100140000, 0x1000, 0x0)		 = 0 0
mprotect(0x100127000, 0x90, 0x1)		 = 0 0
mprotect(0x100141000, 0x1000, 0x1)		 = 0 0
mprotect(0x100127000, 0x90, 0x3)		 = 0 0
mprotect(0x100127000, 0x90, 0x1)		 = 0 0
getpid(0x0, 0x0, 0x0)		 = 37091 0
stat64("/AppleInternal\0", 0x7FFEEFBFE2F0, 0x0)		 = -1 2
csops_audittoken(0x90E3, 0x7, 0x7FFEEFBFDE40)		 = -1 22
proc_info(0x2, 0x90E3, 0xD)		 = 64 0
csops_audittoken(0x90E3, 0x7, 0x7FFEEFBFD6C0)		 = -1 22
sigprocmask(0x3, 0x7FFEEFBFE80C, 0x0)		 = 0x0 0
bsdthread_ctl(0x1000, 0x1, 0x0)		 = 0 0
__pthread_sigmask(0x3, 0x7FFEEFBFE800, 0x0)		 = 0 0
__pthread_kill(0x40B, 0x6, 0x0)		 = 0 0
```

```
$ lldb ./mklittlefs
(lldb) target create "./mklittlefs"
Current executable set to '/Users/anonymous/Downloads/mklittlefs 2/mklittlefs' (x86_64).
(lldb) run
Process 37172 launched: '/Users/anonymous/Downloads/mklittlefs 2/mklittlefs' (x86_64)
Process 37172 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
    frame #0: 0x00007fff6a51833a libsystem_kernel.dylib`__pthread_kill + 10
libsystem_kernel.dylib`__pthread_kill:
->  0x7fff6a51833a <+10>: jae    0x7fff6a518344            ; <+20>
    0x7fff6a51833c <+12>: movq   %rax, %rdi
    0x7fff6a51833f <+15>: jmp    0x7fff6a512629            ; cerror_nocancel
    0x7fff6a518344 <+20>: retq
Target 0: (mklittlefs) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
  * frame #0: 0x00007fff6a51833a libsystem_kernel.dylib`__pthread_kill + 10
    frame #1: 0x00007fff6a5d4e60 libsystem_pthread.dylib`pthread_kill + 430
    frame #2: 0x00007fff6a49f808 libsystem_c.dylib`abort + 120
    frame #3: 0x00000001000b0851 mklittlefs`___lldb_unnamed_symbol3838$$mklittlefs + 369
    frame #4: 0x00000001000b0ebe mklittlefs`___lldb_unnamed_symbol3856$$mklittlefs + 62
    frame #5: 0x000000010000f2b2 mklittlefs`___lldb_unnamed_symbol186$$mklittlefs + 98
    frame #6: 0x00000001000045d7 mklittlefs`TCLAP::CmdLine::missingArgsException() + 307
    frame #7: 0x00000001000047a8 mklittlefs`TCLAP::CmdLine::parse(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >&) + 378
    frame #8: 0x0000000100006b2f mklittlefs`TCLAP::CmdLine::parse(int, char const* const*) + 153
    frame #9: 0x0000000100002c6c mklittlefs`___lldb_unnamed_symbol36$$mklittlefs + 2459
    frame #10: 0x00000001000b32d6 mklittlefs`___lldb_unnamed_symbol3893$$mklittlefs + 6
    frame #11: 0x0000000100000eb4 mklittlefs`___lldb_unnamed_symbol1$$mklittlefs + 52
```

007hacky007, 2020-11-28:
Interestingly it prints some output when I give it some params. However ends up with SIGABRT nonetheless. 
Works fine (without any SIGABRT) when compiled from the source on my local machine.
```
$ ./mklittlefs --help

USAGE:

   ./mklittlefs  {-c <pack_dir>|-u <dest_dir>|-l} [-d <0-5>] [-a] [-b
                 <number>] [-p <number>] [-s <number>] [--] [--version]
                 [-h] <image_file>

Where:

   -c <pack_dir>,  --create <pack_dir>
     (OR required)  create littlefs image from a directory
         -- OR --
   -u <dest_dir>,  --unpack <dest_dir>
     (OR required)  unpack littlefs image to a directory
         -- OR --
   -l,  --list
     (OR required)  list files in littlefs image

   -d <0-5>,  --debug <0-5>
     Debug level. 0 means no debug output.

   -a,  --all-files
     when creating an image, include files which are normally ignored;
     currently only applies to '.DS_Store' files and '.git' directories

   -b <number>,  --block <number>
     fs block size, in bytes

   -p <number>,  --page <number>
     fs page size, in bytes

   -s <number>,  --size <number>
     fs image size, in bytes

   --,  --ignore_rest
     Ignores the rest of the labeled arguments following this flag.

   --version
     Displays version information and exits.

   -h,  --help
     Displays usage information and exits.

   <image_file>
     (required)  littlefs image file

[1]    37272 abort      ./mklittlefs --help
```

earlephilhower, 2020-11-29:
While I believe you, it's not being reproduced in our Mac-based CI on the ESP8266 Arduino or reported by any of our Mac users, so there's nothing I can suggest looking at.  Might be some issue with shared library versions (i.e. a different prototype somewhere), but it's something you'd need to pop into GDB and debug on your own since it's not reproducible.  Good luck!

meticulousCraftman, 2022-12-26:
I am facing the same problem on my machine. I'm using Macbook Air M1 running macOS Monterey (12.4). The macOS release binary does not work on my machine either. But building it from source works perfectly fine.

PR #13 (2020-11-28): Remove stale "test" Makefile target
Fixes #12