earlephilhower/mklittlefs #19: file name too long issue

tsctrl, 2021-06-12:
hi,

i getting error when building image file as below:
unable to open '/data/assets/packages/material_design_icons_flutter/lib/fonts/materialdesignicons-webfont.ttf.gz.
error adding file!

after renaming the file i am able to run the command successfully, however i am not able to rename the file as it was default apps  library path. please help
command i use: mklittlefs.exe -c data -b 4096 -p 256 -s 9699328 mkspiffs.bin

thanks in advance

earlephilhower, 2021-06-12:
Sorry, that's a limitation of the ESP8266 filesystem implementation.  The max file name/directory name is 32 bytes, but your file's name is 34 bytes long (+1 for the terminating \0).  Even if the file was added to the image, the ESP would crash when trying to access it.

badrun-bandi, 2021-06-13:
note and thanks!

tsctrl, 2021-06-13:
noted and thanks!

closed 2021-06-13 by tsctrl