earlephilhower/mklittlefs #22: mklittlefs crashes on symlink

mhaberler, 2021-12-03:
try this (macOS but likely others as well:

````
$ ln -s . internal
$ mklittlefs . -c /tmp/image.bin

/js/jquery/jquery.terminal-2.29.5.min.js.gz
lfs warn:475: No more free space 20
lfs_write error(-28): File system is full.

error adding file!
Error for adding content from jquery!
....
````
I think the best fix would be to (optionally) skip symbolic links

earlephilhower, 2021-12-03:
Good catch, that was definitely not something we'd thought about before.  Replacing `stat` with `lstat` (on non-WIN32 systems) and checking flags should handle things.

Would you like to propose a PR to fix, since you found it and have a solution?

mhaberler, 2021-12-03:
will give it a try

earlephilhower, 2021-12-03:
Great!

As I was thinking about it, it may be a little more complicated.  You'll want to do `stat` to see if it's a directory, and then if do `lstat` to see if it's a symlink to a directory and if so skip.  But you don't want to skill all symlinks (i.e. a linked file), I imagine, just directories.

earlephilhower, 2022-01-19:
Fixed by #23

closed 2022-01-19 by earlephilhower