earlephilhower/mklittlefs #29: Latest stable can't unpack image, bleeding edge can

maxgerhardt, 2022-10-14:
Extension of https://github.com/maxgerhardt/pio-esp32-esp8266-filesystem-downloader/issues/6.

Given [this LittleFS image](https://github.com/maxgerhardt/pio-esp32-esp8266-filesystem-downloader/files/9788256/downloaded_fs_0x300000_0xfa000.zip) that was extracted from an ESP8266, and the latest Windows release from https://github.com/earlephilhower/mklittlefs/releases, unpacking fails (but listing is fine).

```
>C:\Users\Max\Downloads\x86_64-w64-mingw32-mklittlefs-295fe9b\mklittlefs\mklittlefs.exe -b 2048 -p 256 -l downloaded_fs_0x300000_0xfa000.bin
19270   /config.html    Thu Oct 13 08:44:23 2022
965     /set.js Thu Jan 01 00:00:00 1970
256     /settings.bin   Thu Jan 01 00:00:00 1970

>C:\Users\Max\Downloads\x86_64-w64-mingw32-mklittlefs-295fe9b\mklittlefs\mklittlefs.exe -b 2048 -p 256  -u unpack_dir downloaded_fs_0x300000_0xfa000.bin
Assertion failed!

Program: C:\Users\Max\Downloads\x86_64-w64-mingw32-mklittlefs-295fe9b\mklittlefs\mklittlefs.exe
File: littlefs/lfs.c, Line 84

Expression: block < lfs->cfg->block_count
```
In the quest to debug this, I cloned and compiled the repo from source. But then, the created `mklittlefs` can unpack it just fine.

```
>C:\Users\Max\temp\mklittlefs\mklittlefs.exe -b 2048 -p 256 -l downloaded_fs_0x300000_0xfa000.bin
19270   /config.html    Thu Oct 13 08:44:23 2022
965     /set.js Thu Jan 01 00:00:00 1970
256     /settings.bin   Thu Jan 01 00:00:00 1970
Creation time:  Fri Oct 14 15:39:26 2022

>C:\Users\Max\temp\mklittlefs\mklittlefs.exe -b 2048 -p 256 -u unpack_dir downloaded_fs_0x300000_0xfa000.bin
config.html      > ./unpack_dir/config.html     size: 19270 Bytes
set.js   > ./unpack_dir/set.js  size: 965 Bytes
settings.bin     > ./unpack_dir/settings.bin    size: 256 Bytes
```

My questions / requests: Can it *easily* be found out why the stable version crashes upon file extraction, but bleeding edge works? If the latest stable fails to unpack the filesystem, but latest bleeding edge can, can a new stable release be made?

maxgerhardt, 2022-10-14:
Actually I immediately take back the "it unpacks fine". The extracted files contain only unreadable garbage, although a hexdump of the original littlefs image reveals human-readable content.

earlephilhower, 2022-10-14:
I have to admit that I haven't actually used the mklittlefs releases from here, ever, to my knowledge.  I've only done the bundles as part of esp-quick-toolchain or pico-quick-toolchain.

Are you actively using these binaries and not the toolchain bundles above (which build the binaries separately in a docker image w/all the other xtensa/arm GCC bits)?

If not, I will delete these releases.  If so, I could blow away the old releases and do a manual build/publish from the same Dcoekr image/sources that the other toolchains use.

maxgerhardt, 2022-10-15:
Hmm but currently as given, after my correction, neither the stable nor bleeding edge tool cannot unpack the filesystem properly, although good content is visible in the .bin file :/

earlephilhower, 2022-10-15:
Ah, I misunderstood.  Let me look into it...

Just to be 100% clear, the generated images are OK for you?  Only unpacking fails?

Unpacking could have been busted for a long time because I'd say 99.9% of the usage is only generating for uploading to the chips.

maxgerhardt, 2022-10-15:
When I initially created https://github.com/maxgerhardt/pio-esp32-esp8266-filesystem-downloader 2 years ago as a download-and-unpack tool, it worked fine back then with ESP8266 + LittleFS. The filesystem image I referenced however comes from the original issue and from unknown firmware but known toolchain versions. I'll recheck with *my* reference firmware if that is broken too.

maxgerhardt, 2022-10-15:
Well that's weird, my reference example works normally. Prepackaged filesystem uploaded + add a new file dynamically in the firmware code + download again, it unpacks correctly and has the right content.
```
Download was okay: True. File at: C:\Users\Max\temp\pio-esp32-esp8266-filesystem-downloader\downloaded_fs_0x300000_0xfa000.bin
Executing extraction command: "C:\Users\Max\.platformio\packages\tool-mklittlefs\mklittlefs" -b 8192 -p 256 --unpack "unpacked_esp8266_littlefs" "C:\Users\Max\temp\pio-esp32-esp8266-filesystem-downloader\downloaded_fs_0x300000_0xfa000.bin"
hello.txt        > ./unpacked_esp8266_littlefs/hello.txt        size: 27 Bytes
prepared_file.txt        > ./unpacked_esp8266_littlefs/prepared_file.txt        size: 59 Bytes
Unpacked filesystem.
..
>type unpacked_esp8266_littlefs\*

unpacked_esp8266_littlefs\hello.txt

Hello from ESP8266 LittleFS
unpacked_esp8266_littlefs\prepared_file.txt

This is a prepared file, packed + upload to the filesystem.
```

@Miq1 can you upload the firmware you were using? Or minimize it down to only the LittleFS read and writes? Can you lower the `upload_speed` in the `platformio.ini` so that it gets a possibly cleaner dump of the flash and reupload the bin?

Question still remains, that `.bin` file has good looking content and the unpacked one is *completely* broken.

Miq1, 2022-10-15:
@maxgerhardt That will take some time - the firmware is quite big and I need to strip it down a lot. PLS SBY.

Miq1, 2022-10-15:
Okay, looks like I got a shorter version behaving the same.

- the filesystem initially was uploaded with only the ``config.html`` file in it.
- I did an ``erase flash``, ``build file system`` before
- finally I uploaded the sketch enclosed

The sketch is supposed to notice there is no ``settings.bin`` file, then shall write one with the data of the ``settings`` struct. Next the same data will be written into a ``set.js`` file as Javascript statements. The sketch will provide a web server, where you can request the files in question and see these in your browser (like ``http://<your assigned IP>/set.js``)

I can see traces of the three files in the downloaded binary again.

Sketch:
```
/ Copyright 2022 bla foo bar

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <LittleFS.h>
#include <ESP8266WebServer.h>

enum MODE : uint8_t { RUN=0, CONFIG, MANUAL};
MODE mode(RUN);

// Web server definitions
ESP8266WebServer HTMLserver(80);
#define SET_JS "/set.js"
#define CONFIG_HTML "/config.html"
#define DEVICE_HTML "/device.html"

// Choice list for target and sensor devices
enum DEVICEMODE : uint8_t { DEV_NONE=0, DEV_LOCAL, DEV_MODBUS };
enum DEVICECOND : uint8_t { DEVC_NONE=0, DEVC_LESS, DEVC_GREATER };

// Settings data
#define SETTINGS "/settings.bin"
const uint16_t MAGICVALUE(0x4715);
const uint8_t STRINGPARMLENGTH(32);
struct SetData {
  uint16_t magicValue;                   // 0x4712 upon successful initialization
  char deviceName[STRINGPARMLENGTH];     // CV0 Name of this device for mDNS, OTA etc.
  char WiFiSSID[STRINGPARMLENGTH];       // CV1 SSID of local WiFi
  char WiFiPASS[STRINGPARMLENGTH];       // CV2 Password for local WiFi
  char OTAPass[STRINGPARMLENGTH];        // CV3 OTA password
  bool masterSwitch;                     // CV4 Functions on/off main switch
  DEVICEMODE Target;                     // CV7 0=none, 1=local, 2=Modbus
  uint8_t hystSteps;                     // CV5 hysteresis step count 0..15 (16, 1..15)
  uint16_t measuringInterval;            // CV6 time between measurements in seconds
  IPAddress targetIP;                    // CV8..CV11 
  uint16_t targetPort;                   // CV12 target Modbus port number
  uint8_t targetSID;                     // CV13 target Modbus server ID
  struct SensorData {
    DEVICEMODE type;                     // S0:CV14 S1:CV28 0=none, 1=local, 2=Modbus
    IPAddress IP;                        // S0:CV15..CV18 S1:CV29..CV32
    uint16_t port;                       // S0:CV19 S1:CV33 Sensor Modbus port number
    uint8_t SID;                         // S0:CV20 S1:CV34 Sensor Modbus server ID
    bool slot;                           // S0:CV21 S1:CV35 Sensor Modbus slot: false=slot 1, true=slot 2
    DEVICECOND TempMode;                 // S0:CV22 S1:CV36 Sensor temperature condition 0:ignore, 1:<, 2:>
    float Temp;                          // S0:CV23 S1:CV37 Sensor condition temperature value
    DEVICECOND HumMode;                  // S0:CV24 S1:CV38 Sensor humidity condition 0:ignore, 1:<, 2:>
    float Hum;                           // S0:CV25 S1:CV39 Sensor condition humidity value
    DEVICECOND DewMode;                  // S0:CV26 S1:CV40 Sensor dew point condition 0:ignore, 1:<, 2:>
    float Dew;                           // S0:CV27 S1:CV41 Sensor condition dew point value
  } sensor[2];
  DEVICECOND TempDiff;                   // CV42 (S0 - S1) temperature condition 0:ignore, 1:<, 2:>
  float Temp;                            // CV43 (S0 - S1) condition temperature value
  DEVICECOND HumDiff;                    // CV44 (S0 - S1) humidity condition 0:ignore, 1:<, 2:>
  float Hum;                             // CV45 (S0 - S1) condition humidity value
  DEVICECOND DewDiff;                    // CV46 (S0 - S1) dew point condition 0:ignore, 1:<, 2:>
  float Dew;                             // CV47 (S0 - S1) condition dew point value
  uint16_t restarts;                     // number of reboots
[... 216 lines cut ...]
  settings.Dew = -0.5;

  // Run mode?
  if (mode == RUN) {
    // Start WiFi
    wifiSetup(settings.deviceName);
  } else {
    // No, config mode
  }

  // Set up mode independent web server callbacks
  HTMLserver.onNotFound(notFound);
  HTMLserver.enableCORS(true);
  HTMLserver.serveStatic("/", LittleFS, "/");

  // Start web server
  HTMLserver.begin(80);
}

void loop() {
  HTMLserver.handleClient();
}
```

[downloaded_fs_0x300000_0xfa000.zip](https://github.com/earlephilhower/mklittlefs/files/9792193/downloaded_fs_0x300000_0xfa000.zip)

Jason2866, 2022-12-12:
I think it is the same issue as #20

PR #31 (2023-01-29): Automatically grab FS size from image size
Fixes #20
Fixes #29
Fixes #14

The CLI required the user to pass in the FS size, but it seems most people neglected to do this and the default 0x10000 was used, causing LittleFS to truncate most images.

Now require the -s size option on creation, and auto-read the size when doing a list or update operation.  Users can still manually code in a size from the command like if needed.
commit f37ec21e (2023-01-29): Automatically grab FS size from image size

Fixes #20
Fixes #29
Fixes #14

The CLI required the user to pass in the FS size, but it seems most people
neglected to do this and the default 0x10000 was used, causing LittleFS to
truncate most images.

Now require the -s size option on creation, and auto-read the size when doing
a list or update operation.  Users can still manually code in a size from the
command like if needed.
closed 2023-01-29 by earlephilhower