earlephilhower/mklittlefs #38: error: 'sprintf' is deprecated

uztbt, 2023-10-04:
I got the following errors when I tried to build on my M2 Mac.

```
~/g/A/A/t/mklittlefs►make dist                                                                                                                                               3.07s (master) 14:21
c++ -std=gnu++11 -Os -Wall -Wextra -Werror   -Itclap -Iinclude -Ilittlefs -I. -D VERSION=\"0.2.3-62-g975bd0f\" -D LITTLEFS_VERSION=\"v2.5.1\" -D BUILD_CONFIG=\"\" -D BUILD_CONFIG_NAME=\"-generic\" -D __NO_INLINE__ -D LFS_NAME_MAX=32   -c -o main.o main.cpp
main.cpp:459:13: error: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Werror,-Wdeprecated-declarations]
            sprintf(newpath, "%s/%s", path, it.name);
            ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
main.cpp:620:17: error: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Werror,-Wdeprecated-declarations]
                sprintf(newPath, "%s/%s/", lfsDir, ent.name);
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
main.cpp:622:17: error: 'sprintf' is deprecated: This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead. [-Werror,-Wdeprecated-declarations]
                sprintf(newPath, "%s/", ent.name);
                ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
3 errors generated.
make: *** [main.o] Error 1
```

earlephilhower, 2023-10-04:
Yeah, that's `clang` lying and saying it's `gcc` and throwing warnings that GCC doesn't.  (FWIW, MS Visual C started doing that back in the 90s.)

  I suppose it's theoretically possible, but practically I'm not seeing any real cases where `mklittlefs` would have any elevated privileges and be exposed in a manner to really have "security concerns."

In any case, thx for the PR!

closed 2023-10-04 by earlephilhower

uztbt, 2023-10-05:
Wow, I didn't know anything about it. Thank you for teaching me the history🥸

> I suppose it's theoretically possible, but practically I'm not seeing any real cases where mklittlefs would have any elevated privileges and be exposed in a manner to really have "security concerns."

I see!

commit 6e2fa178 (2023-10-04): Merge pull request #39 from uztbt/master

Use snprintf instead of sprintf for security reasons

Fixes #38