earlephilhower/mklittlefs #40: LittleFS image Corrupted with Block Argument

uztbt, 2023-10-04:
## With block parameter

The resulting filesystem is corrupted. Note that `size = 20 * block`.

```
~/g/A/A/temhum-provision►mklittlefs/mklittlefs --create ../temhum/data --page 256 --block 8192 --size 163840  temhumLittleFs.bin                                                            15:38
~/g/A/A/temhum-provision►mklittlefs/mklittlefs -l temhumLittleFs.bin                                                                                                                        15:39
littlefs/lfs.c:4189:error: Invalid block count (20 != 40)
littlefs/lfs.c:1225:error: Corrupted dir pair at {0x0, 0x1}
unable to open directory ''
```

## Without block parameter

The command successfully creates a LittleFS image.

```
~/g/A/A/temhum-provision►mklittlefs/mklittlefs --create ../temhum/data --page 256 --size 1572864 temhumLittleFs.bin                                                                         15:39
~/g/A/A/temhum-provision►mklittlefs/mklittlefs -l temhumLittleFs.bin                                                                                                                        15:40
1844    /AmazonRootCA1AndCA3.pem        Sun Oct  1 13:19:31 2023
1224    /device-certificate.pem.crt     Sun Oct  1 13:44:14 2023
...(omitted)...
Creation time:  Wed Oct  4 06:40:33 2023
```

earlephilhower, 2023-10-04:
The parameters of the FS are not stored in the LittleFS image, you need to supply them on the command line for all operations.

Adding the `--page`  and `--block` options to the `-l` will get you going.

closed 2023-10-04 by earlephilhower

uztbt, 2023-10-05:
Oh, I presumed that the parameters are engraved into the LittleFS image.
As you told me, it worked if I specified the same page and block parameters for both the create and list sub commands!