earlephilhower/mklittlefs #50: Missing libwinpthread-1.dll Error when Running on Windows 11

Tomosawa, 2024-09-03:
Description:

When attempting to run mklittlefs.exe on a Windows 11 system, an error message appears indicating that the libwinpthread-1.dll file is missing. As a result, the application fails to execute.

Steps to Reproduce:

Download release 4.0.0  [x86_64-w64-mingw32-mklittlefs-2f7654f.zip](https://github.com/earlephilhower/mklittlefs/releases/download/4.0.0/x86_64-w64-mingw32-mklittlefs-2f7654f.zip) and extract mklittlefs.exe.
Attempt to run mklittlefs.exe on a Windows 11 machine.
Cannot run the program due to the missing `libwinpthread-1.dll` file. 

This might be an issue with not being statically compiled.

earlephilhower, 2024-09-03:
We actually do build static, but with the latest MinGW (GCC 14?) that's no longer working evidently.
https://github.com/earlephilhower/mklittlefs/blob/2f7654fe808a64ffaae2776344c2d8483e84b7a2/Makefile#L33

I'll try a few more things and worst case package the DLLs with the ZIP and call it done...

earlephilhower, 2024-09-03:
Yup, have to ship the DLL. Fixed in 4.0.1 ZIPs

closed 2024-09-03 by earlephilhower

Tomosawa, 2024-09-04:
Thank you very much for your reply. You can also try this compilation parameter to see if it works: 
``` bash
-static-libgcc -static-libstdc++ -Wl,-Bstatic -lstdc++ -lpthread -Wl,-Bdynamic
```
Note that -lstdc++ comes before -lpthread.

earlephilhower, 2024-09-04:
Thanks, that actually seems to work!

`TARGET_LDFLAGS = -Wl,-static -static-libgcc -static-libstdc++ -Wl,-Bstatic -lstdc++ -lpthread -Wl,-Bdynamic`

I'll get it in the next release and maybe use this same recipe on the Pico-Quick-Toolchain to get the `openocd` and `picotool` running w/o shipping DLLS...