earlephilhower/mklittlefs #58: Bug: LFS_NAME_MAX increase to 255 doesn't appear to have done anything (Cannot create filesystem with object name longer than 32 characters)

pjensen-addco, 2024-12-16:
I'm attempting to create a LittleFS filesystem on an ESP32-S3 using PlatformIO and the ESP-IDF. I need to have a few files with UUIDv4s as the file name, which appears to be tripping up mklittlefs.

When I attempt to "Build Filesystem Image" or "Upload Filesystem Image" I get the following error.
"unable to open (long file name)" into "error adding file!"
[image]

I've attempted to run menuconfig and change the LittleFS parameter CONFIG_LITTLEFS_OBJ_NAME_LEN to 255+, but that does not appear to be changing anything in mklittlefs.

Platformio Packages:
```
PACKAGES:
 - framework-espidf @ 3.50301.0 (5.3.1)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23800.240113 (2.38.0)
 - toolchain-riscv32-esp @ 13.2.0+20240530
 - toolchain-xtensa-esp-elf @ 13.2.0+20240530
```

Minimal Project file included.
[LittleFSTest.zip](https://github.com/user-attachments/files/18156230/LittleFSTest.zip)

earlephilhower, 2024-12-16:
I have no idea what app version Espressif is using for your install (says V2.3 but there never was a release by that number here), but it's not the latest.  We're at V 4.0.1 and it definitely does store long filenames.  I suggest you check with Espressif to see if they're packaging a very old version or if you're using an outdated install...

````
earle@amd:~/src/mklittlefs$ touch tests/12345678901234567890123456789012345678901234567890  # (make a long name file, 50 chars)
earle@amd:~/src/mklittlefs$ ./mklittlefs -c tests -s $((1024 * 1024)) out.fs
/12345678901234567890123456789012345678901234567890
/test_create
/test_create.out
earle@amd:~/src/mklittlefs$ strings out.fs | grep 12345  #  (checking that the full name is there)
12345678901234567890123456789012345678901234567890 
````

closed 2024-12-16 by earlephilhower

pjensen-addco, 2024-12-16:
Are you not the maintainer of this tool?

https://registry.platformio.org/tools/platformio/tool-mklittlefs

earlephilhower, 2024-12-16:
Yes, but Platform.IO asked me to not upload any more packages a couple years ago because of a dispute they were having w/Raspberry Pi (I do the community Pico Arduino core and was publishing there).  It's P.IO's platform so they get to set the rules, we just get to play by them, so there's nothing I personally can do.

You could manually d/l the repo file here and overwrite your old exe, probably.  The command line options should all be the same, only the LittleFS version and configs have changed (modulo bug fixes).

earlephilhower, 2024-12-16:
FYI, P.IO packages of the latest mklittlefs with all the needed JSON/etc. is available at
https://github.com/earlephilhower/pico-quick-toolchain/releases/tag/4.0.1

You should be able to point the tool download to that version in your .ini

pjensen-addco, 2024-12-16:
Oh, I'm so sorry to hear about all that. I hadn't heard about that drama and just spent some time reading up on it all. I can tell that it is a frustrating experience. 

I did download the exe and put it in the folder and it did work right away. I'll try to put that platform in the ini now.

Thank you for all you do for the open source community.