earlephilhower/mklittlefs #62: Any hope of getting a brew formula for this?

JetForMe, 2025-03-18:
Any hope of getting a [brew](https://brew.sh) formula for this? TIA!

earlephilhower, 2025-03-18:
Sorry, the last time I used a Mac was pre-millenium so I'm not really equipped to make one.  I use GH Actions to build the Mac binaries here (and osxcross for my toolchains).

If you're conversant in making brewfiles then maybe you'd like to give it a go?  The relevant lines from the "build release" action are:
https://github.com/earlephilhower/mklittlefs/blob/db0513ade5d4ffb757b0529ce277d0be93e4e46a/.github/workflows/make-release.yml#L70-L98

As you can see, it's just a 1-line `make` and whatever special packaging is required for your pkg system...

earlephilhower, 2025-04-07:
@JetForMe were you able to give it a try?  Is there a brew recipe I should point people to in the README.md?

JetForMe, 2025-04-08:
Oh not yet. Not sure when I'll get to it. I just built it manually.

commit 92dbf11e (2025-09-01): Generate a source brewfile for MacOS

Brew seems to barf on a binary-only Formulat when run against the auditor.
Rewrite the formula to build from source on the target machine.  It's only
a very small number of files and this keeps it all source.

Formula verified on latest brew audit and local install.

Fixes #62
commit 44b30060 (2025-09-01): Generate a source brewfile for MacOS

Brew seems to barf on a binary-only Formulat when run against the auditor.
Rewrite the formula to build from source on the target machine.  It's only
a very small number of files and this keeps it all source.

Formula verified on latest brew audit and local install.

Fixes #62
commit 03d5c3f5 (2025-09-01): Generate a source brewfile for MacOS

Brew seems to barf on a binary-only Formulat when run against the auditor.
Rewrite the formula to build from source on the target machine.  It's only
a very small number of files and this keeps it all source.

Formula verified on latest brew audit and local install.

Fixes #62
commit 2ed30871 (2025-09-01): Generate a source brewfile for MacOS

Brew seems to barf on a binary-only Formulat when run against the auditor.
Rewrite the formula to build from source on the target machine.  It's only
a very small number of files and this keeps it all source.

Formula verified on latest brew audit and local install.

Fixes #62