earlephilhower/mklittlefs #64: lfs error:969: Corrupted dir pair at 81 82

MarcusVDTAVARES, 2025-03-27:
Hello :-)

Can you help with this error I am having with the bin file downloaded from ESP32-S3?

[esp32-s3-devkitc-1-n16r8v.zip](https://github.com/user-attachments/files/19477838/esp32-s3-devkitc-1-n16r8v.zip)

U:\Dropbox\Pessoal\Python Scripts\Ignitron>"C:\Users\marcu\.platformio\packages\tool-mklittlefs\mklittlefs" -u "unpacked_fs" -d 5 -b 4096 -s 1966080 -p 256 "E:\Temp\downloaded_fs_0x210000_0x1e0000.bin"
Debug output enabled
lfs error:969: Corrupted dir pair at 81 82

"C:\Users\marcu\.platformio\packages\tool-mklittlefs\mklittlefs" --version                 
                                                                   
C:\Users\marcu\.platformio\packages\tool-mklittlefs\mklittlefs  version: 0.2.3-35-g943d2f7

U:\Dropbox\Pessoal\Python Scripts\Ignitron>"C:\Users\marcu\.platformio\packages\tool-mklittlefs\mklittlefs" -l -d 5 -b 4096 -s 1966080 -p 256 "E:\Temp\downloaded_fs_0x210000_0x1e0000.bin"               
Debug output enabled
lfs error:969: Corrupted dir pair at 81 82
1789    /12stracoustrummingCN1.json     Sat Mar 01 22:39:56 2025
1727    /AcDc.json      Sat Mar 01 22:39:56 2025
1840    /AintTalkinBoutLoveLARSON.json  Sat Mar 01 22:39:56 2025
1342    /AliceCooper.json       Sat Mar 01 22:39:56 2025
1904    /Anastasia.json Sat Mar 01 22:39:56 2025
1351    /AngryMonkey.json       Sat Mar 01 22:39:56 2025
1843    /BangBang.json  Sat Mar 01 22:39:56 2025
1346    /BendTheFend.json       Sat Mar 01 22:39:56 2025
1337    /BluesArk.json  Sat Mar 01 22:39:56 2025
1822    /BohemianRhapsody.json  Sat Mar 01 22:39:56 2025
1266    /BritishAccent.json     Sat Mar 01 22:39:56 2025
1285    /CleanForBallads.json   Sat Mar 01 22:39:56 2025
1750    /CleanRB500.json        Sat Mar 01 22:39:56 2025
1918    /CleanRB5001.json       Sat Mar 01 22:39:56 2025
1795    /ComfortablyNumbSolo.json       Sat Mar 01 22:39:56 2025
1319    /CountryRoad.json       Sat Mar 01 22:39:56 2025
1328    /CrazyCrue.json Sat Mar 01 22:39:56 2025
1321    /CrazyTrain.json        Sat Mar 01 22:39:56 2025
1308    /DancingInTheRoom.json  Sat Mar 01 22:39:56 2025
lfs error:969: Corrupted dir pair at 81 82
Creation time:  Sat Mar 15 18:20:47 2025

earlephilhower, 2025-04-07:
It means what it says, we're just dumping the LFS internal errors as they come.  So there seems to be an issue with the downloaded binary image.  Make sure you aren't extracting a live filesystem (i.e. ensure you run LittleFS.end() before dumping) so there's no possibility of things like cached writes.  You should also triple check that the parameters (block size, page size, etc.) are all matching what your app used to make the LittleFS filesystem, but if they match there's not much to say or do about it here.

closed 2025-04-07 by earlephilhower

MarcusVDTAVARES, 2025-04-07:
Thank you!
I discovered later that the issue is random, meaning that sometimes it works. I am now guessing that the download speed was corrupting the file sometimes. Thank you anyway for the response.